up-board/up-community #3: Unable to detect i2c device

indoorrobot, 2021-04-26:
Hi, I have updated my upboard to use kernel 5.4.0-1-generic for Ubuntu 18.04 and since then I do not see my i2c device (mlx90641).
When listing the i2c devices using `i2cdetect -l` i see 
```
i2c-3	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-10	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-8	i2c       	i915 gmbus panel                	I2C adapter
i2c-6	i2c       	i915 gmbus ssc                  	I2C adapter
i2c-4	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-11	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-9	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-7	i2c       	i915 gmbus vga                  	I2C adapter
i2c-5	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-12	i2c       	DPDDC-D                         	I2C adapter
```
```
i2cdetect -r -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    
```                     

I have an upboard with the previous kernel (4.4.0-2-upboard) here is looks sorted:

```
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	i2c       	i915 gmbus ssc                  	I2C adapter
i2c-4	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-5	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-6	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-7	i2c       	i915 gmbus vga                  	I2C adapter
i2c-8	i2c       	i915 gmbus panel                	I2C adapter
i2c-9	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-10	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-11	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-12	i2c       	DPDDC-D                         	I2C adapter

```

And I can see the device here where is should be (0X33):

```
i2cdetect -r -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- 33 -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

```

am I missing something?

dorfindoor, 2021-09-19:
I have also encountered the same issue!
Any updates?

CammyTetehJr, 2021-09-20:
There are two i2c pins on the upboard, Pins 3 and 5 for 12c-5 and pins 27 and 28 for 12c-0. On which Pins do you have your 12c device connected to? You can try to see if the device is listed for example if connected to pins 3 and 5 `i2cdetect -r -y 5`

lovtulip, 2022-03-22:
I have also encountered the same issue too!!!!
anyone can figure it out?

lovtulip, 2022-03-22:
Okay, I figured it out. 
first, i connected i2c to pin 1&2, but it's not i2c_5 as they(AAeon) told. 
Sec, There is a very important information but not mentioned in git hub but in manual. 
"Open-drain I2C bus. Needs the pullup resistors to external 3.3V"

So I connected two 10K ohm resistor from 3.3V to SDL and SDA, Then i found my device on I2C_3. Thanks for god. 
Here is manual: 
https://newdata.aaeon.com.tw/DOWNLOAD/MANUAL/UP%20Xtreme%20(UP-WHL01)%20Manual%206th%20Ed.pdf