up-board/up-community #7: Outdated information on "OpenSource UEFI BIOS"

tandasat, 2021-12-11:
https://github.com/up-board/up-community/wiki/Firmware
Information about "OpenSource UEFI BIOS" in this page is no longer valid. When the Intel open source BIOS is flashed, the latest UP Squared will not boot anymore. I already reported this issue at the UP Forum as [OpenSource UEFI BIOS by Intel appears to be unusable now](https://forum.up-community.org/discussion/4806/opensource-uefi-bios-by-intel-appears-to-be-unusable-now), but for posterity, pasting crux of the problem here:

_To be more precise, the board fails to complete FSP initialization and freezes at a black screen. (...). This is was never seen with mine bought about 2 years ago and another friend's one bought just three months ago._

Since the BIOS is no longer usable, I suggest deleting those sentences unless someone can outline instructions to distinguish when that does and does not work.

_Intel made available the Open Source UEFI BIOS with support for UP2. You can find more information and all the necessary instructions from this link: Open Source Firmware UP2_

CammyTetehJr, 2021-12-13:
Thanks for pointing, it out. We will look at it and make the necessary adjustments. I will like to close this issue here and we can continue the conversation on the thread on the [forum](https://forum.up-community.org/discussion/4806/opensource-uefi-bios-by-intel-appears-to-be-unusable-now).

tandasat, 2021-12-15:
I am happy to discuss the issue further on the forum, but unless the change is made, or a resolution is provided, I do not think we should close this for discoverability by others community members.

CammyTetehJr, 2021-12-20:
We have a new reply on the forum, please check.