up-board/up-community #9: Kernel panic - Ubuntu 18.04.05 Installation

CornerStoneF, 2022-03-10:
Hi, 
I wanted to install Ubuntu 18.04 regarding the Wiki-guide, but got an kernel panic error while processing. 
Board: Up Core 01/16 (UPC-CHT01-A20-0116-A11)

See attached error message:
[image]

I created the bootable usb-stick with "balenaEtcher" and the Ubuntu "image creator tool" neither did work.

I hope you can help me. 
Thank you very much.

--
CornerStoneF

CammyTetehJr, 2022-03-15:
Hi,

Can you try to update the BIOS to latest UCR1BM18 and retry installation of ubuntu. You can find the download link [here](https://downloads.up-community.org/download/up-core-uefi-bios-ucr1bm18/)

CornerStoneF, 2022-03-15:
Hi, 

thank you. 
Updating the BIOS worked for me to solve this problem.

closed 2022-03-15 by CornerStoneF

CammyTetehJr, 2022-03-24:
Glad to hear that