up-board/up-community #13: Hailo-8 M2 key not detected on UP-squared 6000

MohitB9, 2022-04-25:
Hello there,
As per Hailo's documentation https://hailo.ai/wp-content/uploads/2021/04/Hailo-8-M.2-key-AE-Data-Sheet-_01_APR_2021.pdf , Hailo-8 M2 accelerator should be visible in lspci list.

But I can it as "Co-processor : Device 1e60:2864 (rev 01)".

Are there any additional settings required apart from getting started wiki?
Setup is Up-square 6000 board, Ubuntu 20.4 with linux headers updated as per https://github.com/up-board/up-community/wiki/Ubuntu_20.04, Hailo-8 M.2 module inserted in slot properly. Can see red LED on the module powered up.

CammyTetehJr, 2022-06-11:
Glad you have it resolved.

zrruziev, 2022-07-08:
> But I can it as "Co-processor : Device 1e60:2864 (rev 01)".

Type this:

zrruziev, 2022-07-08:
> Hello there, As per Hailo's documentation https://hailo.ai/wp-content/uploads/2021/04/Hailo-8-M.2-key-AE-Data-Sheet-_01_APR_2021.pdf , Hailo-8 M2 accelerator should be visible in lspci list.
>
> But I [...]

To update the Hailo string identifier in the PCI IDs list, run this `sudo update-pciids` <br>Then check `lspci`