up-board/up-community #16: Document on Ubuntu 22.04

SauravDharwadkar, 2023-04-09:
There no Document to be found on ubuntu 22.04 installation and in the provided launchpad ppa:up-division/5.4-upboard jammy / ubuntu 22 is missing as well packages are like year old probably missing lot of security patches due to old kernel sources

dawpud, 2023-05-29:
I am new to UP Boards, but I am curious why they want us to download a custom kernel in the first place. I installed Ubuntu 22.04.2 LTS on an UP 4000 edge without needing a custom kernel to have the system work. Granted, I am not using any HATs or GPIO pins. Nonetheless, if you do not need the custom kernel for it to work, then a generic "Linux install" wiki page could be created.

yukoba, 2023-06-28:
For UP Xtreme i12, Ubuntu 22.04 is necessary.